rambo/TinyWire #13: Define TX and RX buffer size

elademir, 2015-07-11:
Hi Rambo, congratulation for your job, your library works fine, thanks!

I have a question, the max value for .available() is TWI_RX_BUFFER_SIZE-1, by default this value is 16-1=15.
I cannot change the value defining TWI_RX_BUFFER_SIZE with 32,64,etc. into a sketch, always gives me a max value of 15. Why this occurs?

A part of my code(the relevant for this issue)

#define TWI_RX_BUFFER_SIZE   ( 64 )
#define TWI_TX_BUFFER_SIZE   ( 64 )                      //This do not works

void receiveEvent(byte howMany)   
  {
  tws_delay(40);                            //waits all data
  reg=TinyWireS.available();        //this value has a max value of 15
    while(TinyWireS.available())
      {
      char c = TinyWireS.receive();        //Flush buffer
      }
  i2c_regs[0]=reg;                        //prepare "reg" value to send into the next request
  reg_position = 0;
  }

rambo, 2015-07-12:
You need to define the constants before including the library so

#define TWI_RX_BUFFER_SIZE ( 64 )
#define TWI_TX_BUFFER_SIZE ( 64 )
#include <TinyWireS.h>

But there might be some issue with the order of defines and the Arduino preprocessor.

In any case that is a horrible way to do whatever you are trying to do (I can't quite figure it out), if you want to transfer text to write out on a display look at https://github.com/HelsinkiHacklab/reactor/blob/master/software/mwdisplay/mwdisplay.ino (it only has a few 7-segment displays but the same idea can be applied to any lenght of string you can fit into RAM).

elademir, 2015-07-12:
Thanks Rambo, it works perfect. In fact the sketch only sends the data recibed packet size to the master. I want to control the attiny for different tasks like a mini DAQ system. I will check your example as guideline for my project.

regards!

closed 2015-07-12 by rambo