rambo/TinyWire #15: All 0xFF

steowens, 2015-10-04:
For some reason I am getting all 0xFF on the master side when calling this library.  I am using ATTiny84 with Rambo TinyWireS, on the slave side.  And using standard Arduino Wire Library on a Arduino Pro Mini as master.

rambo, 2015-10-04:
Post you code from both the slave and master, make sure you have external pull-up resistors.

steowens, 2016-01-17:
Sorry to take so long to get back to you.  We ended up figuring out how to get the project to work, but I honestly cannot remember now which tiny wire library we used.  We did have to revert to sending single character messages though.

variablo, 2016-05-17:
I get also no response. 

- bootloder for ATtiny84 8Mhz is Burned and uploaded
- example attiny85_i2c_slave.ino is compiled for ATtiny84 and uploaded
- for Master i use BusPirate V3.6 with pull-up resistors

If i start the Makro BusPirate "Searching 7bit I2C address space" i get no adress back.
And if I send [9 rrrr] i get
I2C START BIT
WRITE: 0x09 NACK 
READ: 0xFF 
READ:  ACK 0xFF 
READ:  ACK 0xFF 
READ:  ACK 0xFF 
NACK
I2C STOP BIT

rambo, 2016-05-17:
On 17/05/16 16:03, variablo wrote:
> I get also no response. 
> 
> - bootloder for ATtiny84 8Mhz is Burned and uploaded
> - example attiny85_i2c_slave.ino is compiled for ATtiny84 and uploaded
> - for Master i use BusPirate V3.6 with pull-up resistors
> 

Haven't tested that revision but at least my old bus pirate bit-bangs
the I2C protocol and cannot handle clock stretching. Try with something
that has proper hardware I2C master, like Arduino. There's a simple
REPL-style client linked in the examples.

closed 2016-06-20 by rambo