rambo/TinyWire #25: Review for branch:"interrupt_driven"

rshartog, 2016-02-19:
@rambo  

Hi Eero,

I have pushed significant changes into a branch called "interrupt_driven" that I would like you to review (and test if possible). The onRequest() callback function is interrupt driven as of the merge before last, but this change makes the onReceive() callback function also interrupt driven. I've tested it for days using Uno R3, Mega 2560, Micro Pro (Leonardo) as the master device and a Tiny85 as the slave device.  The new TinyWiresS_Stress_Master/_Slave example programs run reliably with zero communication errors. Please read the description of the last commit in the branch because I put more info in there -- especially the part about deprecating the TinyWireS_stop_check() and tws_delay() functions.

I think that the other existing examples should just work, but I have don't have the master program that is meant to accompany them. I can alter the TinyWiresS_Stress_Master for that purpose, but I thought that you might have the original master program used to test these examples.

Please consider whether you even want to merge these changes into the master branch. My opinion is that legacy code built on the current TinyWireS library version [the non-interrupt driven onReceive() callback function] will work with this library interrupt-driven version because an empty version of TinyWireS_stop_check() is still in the library, but I can't say for sure without know how other people have used the library.

Your review and feedback will be appreciated.

--Scott (rshartog)

JasperHorn, 2017-12-19:
These changes seem to have broken the repeated start condition. However, I was able to make it work with a couple of changes. Would you be interested in a pull request with those changes?

(The same changes also solved another problem I was having, but those hard to quantify beyond "the first I2C message after a boot was consistently failing with specific SBC as master".)

rambo, 2017-12-19:
@JasperHorn sure, just send it this way

closed 2018-02-26 by rambo