rambo/TinyWire #27: Advice, or Help with LCD Library for use with TinyWire

ArcAIN6, 2016-07-01:
I've recently been trying to reduce a project down to a single project box. Rather than using my mega, i've been attempting to run a 20x4 I2C LCD and a hall effect sensor on the ATTiny85. Seeing that there are projects all over the place using TinyWireM and the LCD Library, i thought i could do it as well. But it turns out, my LCD has issues. Namely, it appears to be a weird chinese clone. 

The only LCD library that seems to work with it on the mega, is attached below. I'm hoping you can give me some pointers, or help me convert this library to work with TinyWire so i can finish my project. 

Thank you in advance. 
[LiquidCrystal_I2C-Odd.zip](https://github.com/rambo/TinyWire/files/343962/LiquidCrystal_I2C-Odd.zip)

rambo, 2016-07-03:
I haven't really looked at TinyWireM in a looooong while so I can't say much about it. the LCD library looks bog-standard from I2C point of view thus the only thing I can think of is that the clone is not using the actual PCF8574 as IO expander but emulating it somehow and that piece is acting up. Maybe it stretches the clock, maybe TinyWireM can't handle that since the tinys don't have proper TWI(=I2C) hardware like ATMegas do.

ArcAIN6, 2016-07-04:
The chip is stamped PCF8574T and has the NXP logo also etched into the surface. Could be a clone, dunno. 

Thanks for the reply though :)

closed 2016-07-04 by ArcAIN6