rambo/TinyWire #28: #define TWI_RX_BUFFER_SIZE doesn't work

rokenbuzz, 2017-01-25:
I need a bigger RX buffer, so in my main code I have:
`#define TWI_RX_BUFFER_SIZE ( 64 )`
as the first line, before any #includes.
But it doesn't make a difference.  I see in usiTwiSlave.h:
```
#ifndef TWI_RX_BUFFER_SIZE
#define TWI_RX_BUFFER_SIZE  ( 16 )
#endif
```
So it seems like I should be able to define my own TWI_RX_BUFFER_SIZE in my main code first, and its value would get used.  It doesn't, the code still has 16 byte TX buffer.
Instead if I change the #define in usiTwiSlave.h to 64 then my code runs fine.  But it seems wrong to change a library file like that.
But maybe I have to in this case.  I'm not absolutely sure #defines carry over to other code, but I thought #include pretty much just gets replaced with the actually header file.

An advice?
Buzz

rambo, 2017-01-25:
Yeah there's some weird scoping problem there, never figured it out (to be fair didn't spend too much time looking at it either). The easy way is to just modify the library :( if you can figure out what the scoping issue is please tell here.

rokenbuzz, 2017-01-25:
I now understand that #defines do not leave the scope of the source file.  They can be pulled in from header files though.
From keil.com/support/docs/2589.htm:
> The scope of #define is limited to the file in which it is defined. So, #defines which are created in one source file are NOT available in a different source file.
> 
> Typically, #defines which are shared between multiple files are stored in a header file (*.h) which is included in each source file that requires the #define.

So it is what it is.  I'll just modify the value in my copy of your code.
Thanks for all the work on this.  This code really enabled my project.

closed 2017-01-25 by rambo