rambo/TinyWire #30: i2c slave example

Toffe87, 2017-03-05:
Hello.

I've hooked my attiny with i2c up to my raspbery pi. 

The example contains theese registers:
    0xDE, 
    0xAD, 
    0xBE, 
    0xEF, 

When i try to get register 0x00 i get 0x5e which should be 0xDE

This happends to all registers, seems like they revolv on 0x7f back to 0x00. Setting one register to 0xFF returns 0x00 when read. Setting to 0x7F returns 0x00, Setting to 0x80 returns 0x01

Any idea on what i am doing wrong or if there is a bug?

Toffe87, 2017-03-05:
Seems like changing bootloader to 1Mhz fixed the problem. My question now is are it stable now? Or am i missing something?

rambo, 2017-03-06:
See https://github.com/rambo/TinyWire#note-about-reliable-communication 

if the ATTiny is running at 1MHz instead of 8MHz it's definitely going to be a lot slower, in fact I wonder how it can work at all at such low speeds.

RPi has known bugs in it's I2C hardware http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html *increasing* the clock speed of the attiny has helped some people

closed 2017-09-04 by rambo