rambo/TinyWire #33: Support for ATmega8

beicnet, 2017-05-30:
Hi,

Can you please make a support for ATmega8 MCU too?

Kind regards,
Viktor

rambo, 2017-05-30:
ATMega8 has proper TWI hardware, use https://github.com/rambo/i2c

closed 2017-05-30 by rambo

beicnet, 2017-05-30:
But how I see, it's not compatible with existing "Wire.h" and "TinyWireS.h" library, or?!

Basically I wanted to be like for the ATtiny85, to be shrinked (to be smaller that the existing Wire.h library)

Thank you!

Regards

rambo, 2017-05-30:
It's not directly API compatible, OTOH the Wire.h API is horrid... That I2C master library has much better control over things (it also should compile to smaller size, having less abstractions getting in the way).

Ah, but I just realised we're talking about *slave* mode here... nevermind that *master* library.

Anyway adding support for completely different hardware (USI vs TWI) will only serve to make this library bloated and that's the last thing we want. If Wire.h compiles to too-large binary you can always just use the TWI hardware directly, the ATMega8 datasheet has all the details and it's not hard  to use.

beicnet, 2017-05-30:
Aha, okay! Thank you! ;)