rambo/TinyWire #38: Compilation fails when included through Arduino IDE "Include Library" menu

wesleycha, 2017-11-20:
When the library is included using the "Include Library" menu in the latest Arduino IDE, both the TinyWireS.h file and the usiTwiSlave.h file are included by default. This makes compilation fail until the user comments out the line including "usiTwiSlave.h". Is there a way to revise this behaviour so that the IDE only includes TinyWireS.h by default?

[image]

[image]

[image]

[image]

[image]

While commenting this line out is a simple process, the confusing error message and lack of instructions can confuse many inexperienced users such as myself.

rambo, 2017-11-27:
I'm not sure why Arduino IDE even does that, maybe it includes all `.h` files or something.

Can you copy the full error text from the compilation, this might be solved with an `#ifdef` guard.

wesleycha, 2017-11-29:
The error text is as follows:
```
Archiving built core (caching) in: /tmp/arduino_cache_539456/core/core_attiny_avr_ATtinyX5_cpu_attiny85,clock_internal8_6667d93d01bb9f3f70387db375dc94fa.a
libraries/TinyWireS/TinyWireS.cpp.o (symbol from plugin): In function `usi_onReceiverPtr':
(.text+0x0): multiple definition of `usi_onReceiverPtr'
sketch/test.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/TinyWireS/TinyWireS.cpp.o (symbol from plugin): In function `usi_onReceiverPtr':
(.text+0x0): multiple definition of `usi_onRequestPtr'
sketch/test.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/TinyWireS/TinyWireS.cpp.o (symbol from plugin): In function `usi_onReceiverPtr':
(.text+0x0): multiple definition of `_onTwiDataRequest'
sketch/test.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny25/45/85.
```

per1234, 2018-02-08:
You're correct that by default the Arduino IDE adds includes for all .h files in the library's root source folder. I can think of two solutions:
1. Move usiTwiSlave.h and usiTwiSlave.c to the utility subfolder. From the [Arduino library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#source-code) (since this library is in 1.0 format instead of 1.5 format replace "src/" in the below quote with "/utility"):
  >in general, the only header files in the root src/ folder should be those that you want to expose to the user's sketch and plan to maintain compatibility with in future versions of the library. Place internal header files in a subfolder of the src/ folder.
2. Add a library.properties file that contains the line:
```
includes=TinyWireS.h
```
That will cause only the include of TinyWireS.h to be added to the sketch but this feature was only added in Arduino IDE 1.6.10 so the problem will still occur with older IDE versions. Reference: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format