rambo/TinyWire #41: Setting Slave Address over 0x3F

patrickpoirier51, 2017-12-28:
I can get stable reading when Slave Address is < 0x40 with this code
void setup() {
pinMode(Rx, INPUT);
pinMode(Tx, OUTPUT);
mySerial.begin(9600);
TinyWireS.begin(I2C_SLAVE_ADDR); 
TinyWireS.onRequest(requestEvent); 
}
void requestEvent()
{  
  SendCount++; //increment this
  switch (SendCount)
  {
    case 1:
      sendByte = distLow;
      break;
    case 2:
      sendByte = distHigh;
      SendCount=0;
      break;
  }
  TinyWireS.send(sendByte);
}

When Address is => 0x40 , its spitting garbage.

Wild guess:
Seems to be an issue with the write bit being the 7th bit (instead of 8)

rambo, 2018-01-02:
The write bit is not the issue, but I think I've heard of this problem before but never had any time to really look into it.

anderstogern, 2018-11-23:
Hi there and thanks for this great library!

I can confirm this issue.

For a long period of time (about a year) I have been able to successfully use slave address 0x50 for a slave that only returned a single byte of data and did not receive anything (no receiveEvent defined).

Recently, I had the need to change the slave to return 3 bytes and receive a single byte. Changing the code broke the communication and it was only after several nights of frustration that I discovered this issue and gave changing the address (to 0x20) a try, which immediately fixed the problem.

For good measure, the slave is an AtTiny84 and the master an Adafruit Huzzah ESP8266 (not the Feather one).

Cheers!