rambo/TinyWire #42: Max bus speed?

ludiazv, 2018-02-25:


rambo, 2018-02-26:
I haven't tried it, there's always a chance but it might not be too reliable. If you can adjust the master clock arbitrarily see how high you can go with the 8MHz clock and then make educated guesses.

ludiazv, 2018-02-26:
Thanks Rambo. I will try to do some tests a different speeds. If they are interesting I will report my findings. 

On the other hand,  I saw also that you merged a fully interrupt driven support, so no more need for pooling for star condition. It would be great if you can add tags to the versions a brief change log. This will help libray user to indentify the version used, specially using automated library managers such platformio.

dontsovcmc, 2018-04-11:
That is the maximum  I2C Slave  speed for Attiny85 running on 1Mhz internal oscillator?

ludiazv, 2018-04-11:
Hi,
  After several test I've managed to comunicate at 100kHz and 400kHz but running at 8Mhz and 16Mhz. I have not tested at 1Mhz. The problem is not CLK frequency but time between each transactions. Adding a small delay(few us) btw i2c transactions give time to the tiny to process the request without problems and be able to detect the next stop condition. So in principle its possible to to implement the slave but the master needs to be nice.

For the tests I have used a RPI as master that is problematic but is faster than a typical mcu.

  An important feedback for @rambo that was unable to make it work with the new version based on IRQ but with the old one that require polling of stop condition. I don not know why but stopcondition is not detected. 

Regards,

rambo, 2018-04-11:
There's a bug (or missing feature if you're nice) in the hardware and the I2C STOP condition simply does not have an interrupt vector on the ATTiny USI hardware (at least back in the revisions when I last looked at the documentation and I'd be very surprised if it got "fixed")

As for @ludiazv I'm fairly sure 1MHz  clock will not work except at very slow bus speeds maybe and the master definitely must support "clock stretching" (where slave holds the clock down and master MUST wait for it to come back up before driving it back down for the next bit) properly. Something which RPi I2C master for example does not do correctly.

dontsovcmc, 2018-04-11:
Here is a communication between Attiny85 1Mhz int. as slave with ESP8266 as master. 
```
2018-04-11 02:56:58,787 - root - INFO - handle_data (134): 78 00 01 40 1e 00 01 00 1c 05 eb 2c 43 0a 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
2018-04-11 03:27:29,241 - root - INFO - handle_data (134): 78 00 01 40 1e 00 01 00 1c 0b eb 2c 43 0a 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 01 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 01 00 02 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
2018-04-11 03:57:59,629 - root - INFO - handle_data (134): 78 00 01 40 1e 00 01 00 1c 0b eb 2c 43 0a 02 00 01 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
Attiny85 1800seconds sleeping with watchdog is ~1830seconds. So there is a ~1.6% shift.