rambo/TinyWire #48: When to use TinyWire vs Wire in ATTinyCore (1.1.3+)?

lowell80, 2019-12-19:
For someone who's writing a fresh I2C slave implementation what are the pros/cons of using TinyWireS vs the (now) built-in transparent support for using Wire (as of ATTinyCore 1.1.3 and later)?  Are there unique features, device support, or API advantages to using TinyWire not covered by ATTinyCore?

I'm trying to make a simple infrared receiver that relays IR codes over the I2C bus to another arduino controlling some WS2812s via FastLED.  But beyond my immediate use case, I realized this may be a question others would be interesting in as well.

I was under the impression that TinyWire was the only (or go-to) option until I stumbled upon this: 
 https://github.com/SpenceKonde/ATTinyCore/blob/master/README.md#i2c-support

> I2C functionality can be achieved with the hardware USI. As of version 1.1.3 this is handled transparently via the special version of the Wire library included with this core. 

I see that that ATTiny85 is supported, and I confirmed that I could compile the `slave_sender` example, but haven't looked look any further.

Thanks!

rambo, 2019-12-22:
I have not tested the new ATTinyCore so can't really say anything about the relative merits.

znoxx, 2020-01-30:
I think I can answer here -- probably it will not work. At least my project failed to work after TinyCore update to latest version. Previously it worked ok. I had to switch to "built-in" Wire lib implementaiton in TinyCore itself.